RogueMaster/flipperzero-firmware-wPlugins #705: FBT ERRORS

cokyrain, 2023-07-14:
### Describe the bug.

```
	CC	applications/external/continuity/continuity.c
	CC	applications/external/counter/counter.c
applications/external/counter/counter.c:7:10: fatal error: Counter_icons.h: No such file or directory
    7 | #include <Counter_icons.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
scons: *** [build/f7-firmware-C/.extapps/counter/counter.o] Error 1

********** FBT ERRORS **********
build/f7-firmware-C/.extapps/counter/counter.o: Error 1
Error: Process completed with exit code 2.

```

### Reproduction

Error from commits on Jul 14, 2023， nothing abnormal before

ESurge, 2023-07-14:
This doesn't happen when compiling on Windows

I tried on Ubuntu and it doesn't work. Therefore, I pushed a fix for this but it hasn't been merged yet.

Check out https://github.com/RogueMaster/flipperzero-firmware-wPlugins/pull/704/commits/44ac1d72606f6e8d491ec3292a8f468a35694bb2 for the fix if you don't want to wait.

cokyrain, 2023-07-14:
Thank you very much

closed 2023-07-14 by RogueMaster