RogueMaster/flipperzero-firmware-wPlugins #706: CLI-GUI Bridge Chat Command Missing

lp36000, 2023-07-14:
### Describe the bug.

The chat command is gone from the CLI-GUI bridge

### Reproduction

Just open the cli-gui bridge run the chat command and it does not exist from the flipper itself or from a serial terminal

### Target

Firmware version: 420 0.86.3 (744fe0c8 built on 14-07-2023)

### Logs

```Text
Firmware version: 420 0.86.3 (744fe0c8 built on 14-07-2023)

>: chat
`chat` command not found, use `help` or `?` to list all available commands
>: ?
Commands available:
!                             led
?                             loader
bt                            log
crypto                        nfc
date                          onewire
device_info                   power
factory_reset                 ps
free                          rfid
free_blocks                   source
gpio                          src
help                          start_rpc_session
i2c                           storage
ikey                          sysctl
info                          update
input                         uptime
ir                            vibro
```

RogueMaster, 2023-07-15:
This was fixed recently by ESurge

RogueMaster, 2023-07-15:
Next release will have it fixed

closed 2023-07-15 by RogueMaster