RogueMaster/flipperzero-firmware-wPlugins #711: Shutdown on Idle affects update screen

DarkEnder4583, 2023-07-18:
### Describe the bug.

The Shutdown on Idle feature counts the time spent on the update screen as idle, resulting in failed updates

### Reproduction

1. Set “Shutdown on Idle” to 15 min
2. Connect to Flipper via Mobile App
3. Begin update process
4. (Hope it takes longer than 15 minutes to update)
5. Flipper shuts down and the update will fail

RogueMaster, 2023-07-18:
1. Set “Shutdown on Idle” to 15 min <--- Don't do this part and you won't run into this issue 😓

closed 2023-07-18 by RogueMaster

DarkEnder4583, 2023-07-18:
Well, I figured that much, and that is what I did to work around the bug. Just thought you wouldn't want to settle for a workout, but aiight.