RogueMaster/flipperzero-firmware-wPlugins #717: Flipper Crashes when not in use.

tkbstudios, 2023-07-23:
### Describe the bug.

When I don't use my flipper anymore, after a while, it crashes and reboots.
The error message as follows:
Flipper Crashed and Rebooted
ST(R) GoPro(R) HardFault

### Reproduction

1. Switch on
2. Don't use it for a while
3. Look at the screen after about 6-7h

### Target

0.87.1[19-07-2023]

mriliu, 2023-07-23:
Same problem

RogueMaster, 2023-07-24:
Yes this is a known issue with sleep mode that OFW is working on. For the time being, set you sleep mode to Legacy to avoid it

closed 2023-07-24 by RogueMaster

tkbstudios, 2023-07-24:
Ok, I just updated my flipper with the RM firmware (latest) and it was fixed, so idk 🤷

mriliu, 2023-07-24:
Weird, I have the latest firmware and it still happened for me.

tkbstudios, 2023-07-25:
ok, thanks