RogueMaster/flipperzero-firmware-wPlugins #743: Seader app not working (works on original firmware)

Antiklesys, 2023-09-07:
### Describe the bug.

When reading an iclass SE card using the Seader app in Roguemaster it fails.
Using the original firmware + the seader app installed from the mobile app it works flawlessly.
I spoke with the creator of the Seader app and it seems the error log is throwing errors that were fixed in the latest release.
I suspect the seader app code in roguemaster is stale compared to latest release.

### Reproduction

1- Install roguemaster firmware fresh
2- add the sam+nard reader to the flipper zero
3- scan an iclass SE card with the seader app using the picopass function
4- errors out
5- install original firmware
6- connect to mobile phone via bluetooth and install the seader app from the store
7- perform steps 2 &3 and it will read the card correctly

RogueMaster, 2023-09-07:
1) New Patreon releases have the latest version available with fixes and additional features, are you on the public release build?
2) The weekly public release will have the latest version. So if you aren't on the Patreon build, a new release will be available within the week.
3) Seader issues would usually be in the Seader repo, however, this issue seems to be more an issue with a past release that is currently available in the public RM release.

RogueMaster, 2023-09-07:
Are you on the latest public release @Antiklesys ? Please confirm the build date and hash for your build.

closed 2023-09-07 by RogueMaster

Antiklesys, 2023-09-07:
@RogueMaster I'm on the latest public release

RM0905-0900-0.90.2-95c6220

Antiklesys, 2023-09-07:
Build date is 2 days ago.