RogueMaster/flipperzero-firmware-wPlugins #744: RGB Backlight Freeze

5oAwes0me, 2023-09-09:
### Describe the bug.

I'm very excited for the RGB Backlight Animation feature. Unfortunately when i right click on static to change it to animation the RGB turns orange and my Flipper freezes. Is there a quick fix for this ? 

CFW version: 0.90.2 

### Reproduction

1. CFW settings
2. Misc
3. Screen 
4. LCD Style        Static >
5. Right click 
6. Freeze with RGB turning orange

RogueMaster, 2023-09-09:
Thanks for your report, we are looking into this issue...

ESurge, 2023-09-10:
I pushed a fix for this in PR #745 

When it's merged this issue should be resolved.

5oAwes0me, 2023-09-10:
> I pushed a fix for this in PR #745
> 
> When it's merged this issue should be resolved.

That's Awesome, thanks for the great work.

closed 2023-09-10 by 5oAwes0me