RogueMaster/flipperzero-firmware-wPlugins #760: external antena for the sub-ghz radio

92firebird, 2023-10-02:
### Describe the bug.

I can no longer select an external antena for the sub-ghz  radio i thought maybe it was my settings but it is not i am also using the latest update of the firmware i changed it to the flipper brand firmware and it works

### Reproduction

The ext power amp is on (has no effect on the option to select antena) 

### Anything else?

i dont have any logs to share as i am only a month or 2 into using my flipper and i dont think there would be any anyway

RogueMaster, 2023-10-02:
Make sure default is selected in CFW Settings => Protocols... and the device needs to be plugged in for the option to come up...

closed 2023-10-02 by RogueMaster

92firebird, 2023-10-02:
thank you an i should have checked the closed bugs my apologies