RogueMaster/flipperzero-firmware-wPlugins #768: What was i banned for?

SylveonDeko, 2023-10-10:
### Describe the enhancement you're suggesting.

I dont know what spam you're talking about. If you are talking about the 2 emotes i sent in weeb central im sorry. otherwise if its something else ive been too busy this morning to even be on discord.

RogueMaster, 2023-10-10:
You were banned for trying to DDOS the site and submitting my email to spam senders

closed 2023-10-10 by RogueMaster

SylveonDeko, 2023-10-10:
That wasn't me tho. I was literally busy fixing cameras today. Last night I
was making a plate number API to check if a car is supposed to be parked
where it's parked.

RogueMaster, 2023-10-10:
You may not have submitted my email for spam, but it was the toxic camp you were supporting when you were trying to hack my site that did it, so the ban can remain for now