RogueMaster/flipperzero-firmware-wPlugins #772: StatusBar drawn on menus

LeeroysHub, 2023-10-15:
### Describe the bug.

If Clock is enabled on desktop, it draws on power menu and up arrow shortcut menu.

I added this code to gui.c to fix.

static void gui_redraw_status_bar(Gui* gui, bool need_attention) {
    ViewPortArray_it_t it;
    uint8_t left_used = 0;
    uint8_t right_used = 0;
    uint8_t left_slim_used = 0;
    uint8_t right_slim_used = 0;
    uint8_t top_used = 0;
    uint8_t width;

    //We dont want to draw the status bar if its hidden.
    if(gui->hide_statusbar_count > 0) {
        return;
    }

### Reproduction

1. Enable Clock
2. Hold back t power off and view clock!

RogueMaster, 2023-10-15:
Thank you! Will add

closed 2023-10-15 by RogueMaster

Hellblocc600, 2023-10-15:
> <RogueMaster/flipperzero-firmware-wPlugins/issue/772/issue_event/10658483521
> @github.com>

Quick question does the newest update has android BLE spam

LeeroysHub, 2023-10-15:
Yes, the Apple BLE Spam app is included. There is no Android BLE Spam, just Apple.

AC9DJ, 2023-10-16:
There is a new BLE app on the Xtreme firmware that includes Android and Windows.
Talking Sasquatch covered it in. a video earlier today.
https://youtu.be/d8cSKNmBwX4?si=tSC2e72S7x2n073F

Sent with [Proton Mail](https://proton.me/) secure email.

LeeroysHub, 2023-10-16:
Yes, you are right! Sorry about that. Its psrt of XFW, but Im sure RogueMaster will add it to RM very soon!

https://youtu.be/d8cSKNmBwX4?si=tSC2e72S7x2n073F

RogueMaster, 2023-10-16:
New BLE Spam app is in the latest RM build and coming to Public Release tonight!