RogueMaster/flipperzero-firmware-wPlugins #992: issue NFC -> extra actions -> Unlock SLIX-L

heckpiet, 2023-10-30:
### Describe the bug.

can't use the app to write the file on the SD card.

[image]

https://github.com/RogueMaster/flipperzero-firmware-wPlugins/assets/6371541/c2e4110f-c0b3-4302-90c5-38472412e53c

### Reproduction

1. switch on
2. navigate to NFC
3. extra actions
4. Unlock SLIX-L
5. place it on a Tonie
6. error message "Can not save key file"

RogueMaster, 2023-10-30:
NFC Refactor will come, if it works there, it will work...

closed 2023-10-30 by RogueMaster

heckpiet, 2023-11-02:
still a issue in Version
RM1031-1757-0.93.1-ef656e6

RogueMaster, 2023-11-02:
> still a issue in Version
> RM1031-1757-0.93.1-ef656e6

And that version says in it's readme that the NFC Refactor is deferred... if the issue is still present once the refactor gets pushed in then open a ticket with OFW since our NFC will be the same as theirs...

heckpiet, 2023-11-02:
thanks for the update

disintegr8te, 2023-12-11:
OFW has it fixed in actual Release

RogueMaster, 2023-12-11:
It is also fixed in the RC release

disintegr8te, 2023-12-11:
> It is also fixed in the RC release

Yes, Tested the RC , reading them Works fine there.
But the Unlock Function is gone under Special Operations.

RogueMaster, 2023-12-11:
Yes, this is why my main release remains on the old NFC until the new NFC is no longer a regression