RogueMaster/flipperzero-firmware-wPlugins #1002: In left-handed mode

alucardzlh, 2023-12-01:
### Describe the bug.

Hello, I found that when the firmware is switched to left-hand mode, the left and right buttons will be reversed in the wrong direction. Version number: RM1130-1445-0.93.9-d9db571

### Reproduction

1. On the home page 
2. Press up to enter the quick panel 
3. Click the left-hand mode button in the upper left corner

### Target

RM1130-1445-0.93.9-d9db571

BKBobbyDigital, 2024-01-21:
Noting this is still an issue (buttons inversed when in lefty mode) in 0.93.9 18-01-2024.

Techno3600, 2024-04-14:
Just wanted to chime in and say this is still an issue. I really liked lefty mode on the default firmware, and I'd really like to be able to use it with RogueMaster!

RogueMaster, 2024-04-14:
Please do submit a PR if you can find a solution.  We don't need any useless chimes unless it's a solution. I've spent enough time investigating this minor issue.

Techno3600, 2024-04-18:
> Please do submit a PR if you can find a solution. We don't need any useless chimes unless it's a solution. I've spent enough time investigating this minor issue.

Hi! I don't code in this language, so I can't submit a PR. That would be the job of the people who maintain the project/applicable contributors.

This is not a minor issue. As a lefty, I found lefty mode on the default software to be extremely helpful in using the Flipper, and having to switch back to righty mode just to access the features of this software is frustrating. Your software/plugins are great (and I love using it) but it requires converting back to righty mode, which could be easily fixed by just inverting the arrow keys in the software.

aryn832, 2024-04-27:
https://github.com/RogueMaster/flipperzero-firmware-wPlugins/pull/1063

RogueMaster, 2024-04-27:
Confirm with me on Discord your ETH address and it shall be sent.

closed 2024-04-27 by RogueMaster