RogueMaster/flipperzero-firmware-wPlugins #1007: CFW Settings Error

aoshea83, 2023-12-14:
### Describe the bug.

Issue with 0.93.9 no animations update. When selecting \CFW Settings\Interface\Desktop the flipper crashes and reboots due to a null pointer deference

### Reproduction

1. Select "CFW Settings"
2. Select "Interface"
3. Select "Desktop"
4. Cry

### Logs

```Text
1333 [APP] qFlipper version 1.3.3 commit bfce851d 2023-11-10T09:40:31
1333 [APP] OS info: Windows 11 Version 23H2 11 10.0.22635 Qt 6.4.2
1364 [DBG] Trying serial port FLIP_R0WTUMZ at \\.\COM5
1364 [DBG] Using  serial port FLIP_R0WTUMZ at \\.\COM5
1380 [RPC] Starting RPC session...
1431 [RPC] RPC session started successfully.
1431 [RPC] (1) System Protobuf Version START
1504 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
1531 [DBG] Detected protobuf version: 0.21
1531 [RPC] (1) System Protobuf Version SUCCESS
1544 [RPC] (2) Property Get START
1599 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
1699 [RPC] (2) Property Get SUCCESS
1700 [RPC] (3) Storage Info @/ext START
1764 [RPC] (3) Storage Info @/ext SUCCESS
1764 [RPC] (4) Storage Stat @/ext/Manifest START
1817 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
1817 [RPC] (5) System Get DateTime START
1898 [DBG] Flipper time skew is -1015 milliseconds
1898 [RPC] (5) System Get DateTime SUCCESS
1898 [RPC] (6) System Set DateTime START
1980 [RPC] (6) System Set DateTime SUCCESS
1980 [RPC] Stopping RPC session...
1988 [RPC] Attempting to unload protobuf plugin...
1988 [RPC] Unloaded protobuf plugin.
1988 [RPC] RPC session stopped successfully.
1988 [REG] Registering the device
1988 [DEV] Version: unknown commit: 155e4e9f radio: 1.17.3
1988 [BKD] Current device changed to "R0wtumz"
2099 [RPC] Starting RPC session...
2214 [RPC] RPC session started successfully.
2214 [RPC] (1) Gui Start ScreenStream START
2282 [RPC] (1) Gui Start ScreenStream SUCCESS
4915 [RPC] (2) Storage Info @/ext START
5000 [RPC] (2) Storage Info @/ext SUCCESS
5006 [RPC] (3) Gui Stop ScreenStream START
5098 [RPC] (3) Gui Stop ScreenStream SUCCESS
7163 [RPC] (4) Storage List @/ext START
7298 [RPC] (4) Storage List @/ext SUCCESS
11816 [RPC] (5) Storage List @/ext/update START
11920 [RPC] (5) Storage List @/ext/update SUCCESS
15564 [RPC] (6) Storage List @/ext/update/RogueMaster START
15643 [RPC] (6) Storage List @/ext/update/RogueMaster SUCCESS
24600 [UTL] Upload 1 entry @R0wtumz START
24618 [RPC] (7) Storage Write @/ext/update/RogueMaster/RM1207-2110-0.93.9-a815d93-NoAnim.zip START
70404 [RPC] (7) Storage Write @/ext/update/RogueMaster/RM1207-2110-0.93.9-a815d93-NoAnim.zip SUCCESS
70404 [UTL] Upload 1 entry @R0wtumz SUCCESS
70404 [RPC] (8) Storage List @/ext/update/RogueMaster START
70485 [RPC] (8) Storage List @/ext/update/RogueMaster SUCCESS
115206 [RPC] (9) Storage Remove @/ext/update/RogueMaster/RM1207-2110-0.93.9-a815d93-NoAnim.zip START
115274 [RPC] (9) Storage Remove @/ext/update/RogueMaster/RM1207-2110-0.93.9-a815d93-NoAnim.zip SUCCESS
115274 [RPC] (10) Storage List @/ext/update/RogueMaster START
115362 [RPC] (10) Storage List @/ext/update/RogueMaster SUCCESS
135957 [UTL] Upload 1 entry @R0wtumz START
135978 [RPC] (11) Storage MkDir @/ext/update/RogueMaster/RM1207-2110-NoAnim START
136127 [RPC] (11) Storage MkDir @/ext/update/RogueMaster/RM1207-2110-NoAnim SUCCESS
136142 [RPC] (12) Storage Write @/ext/update/RogueMaster/RM1207-2110-NoAnim/firmware.dfu START
141041 [RPC] (12) Storage Write @/ext/update/RogueMaster/RM1207-2110-NoAnim/firmware.dfu SUCCESS
141041 [RPC] (13) Storage Write @/ext/update/RogueMaster/RM1207-2110-NoAnim/radio.bin START
141989 [RPC] (13) Storage Write @/ext/update/RogueMaster/RM1207-2110-NoAnim/radio.bin SUCCESS
141989 [RPC] (14) Storage Write @/ext/update/RogueMaster/RM1207-2110-NoAnim/resources.tar START
238278 [RPC] (14) Storage Write @/ext/update/RogueMaster/RM1207-2110-NoAnim/resources.tar SUCCESS
238279 [RPC] (15) Storage Write @/ext/update/RogueMaster/RM1207-2110-NoAnim/splash.bin START
238343 [RPC] (15) Storage Write @/ext/update/RogueMaster/RM1207-2110-NoAnim/splash.bin SUCCESS
238343 [RPC] (16) Storage Write @/ext/update/RogueMaster/RM1207-2110-NoAnim/update.fuf START
238427 [RPC] (16) Storage Write @/ext/update/RogueMaster/RM1207-2110-NoAnim/update.fuf SUCCESS
238444 [RPC] (17) Storage Write @/ext/update/RogueMaster/RM1207-2110-NoAnim/updater.bin START
239430 [RPC] (17) Storage Write @/ext/update/RogueMaster/RM1207-2110-NoAnim/updater.bin SUCCESS
239430 [UTL] Upload 1 entry @R0wtumz SUCCESS
239438 [RPC] (18) Storage List @/ext/update/RogueMaster START
239624 [RPC] (18) Storage List @/ext/update/RogueMaster SUCCESS
252327 [RPC] Serial connection was lost.
252344 [RPC] Stopping RPC session...
252346 [RPC] Attempting to unload protobuf plugin...
252347 [RPC] Unloaded protobuf plugin.
252347 [RPC] RPC session stopped successfully.
267975 [REG] Device disconnected: VID_0x483:PID_0x5740
267975 [BKD] Last device was disconnected
292959 [REG] Detected new device: VID_0x483:PID_0x5740
292971 [DBG] Trying serial port FLIP_R0WTUMZ at \\.\COM5
292971 [DBG] Using  serial port FLIP_R0WTUMZ at \\.\COM5
292972 [RPC] Starting RPC session...
293032 [RPC] RPC session started successfully.
293032 [RPC] (1) System Protobuf Version START
293131 [DBG] Detected protobuf version: 0.20
293131 [RPC] (1) System Protobuf Version SUCCESS
293131 [RPC] (2) Property Get START
293287 [RPC] (2) Property Get SUCCESS
293296 [RPC] (3) Storage Info @/ext START
293348 [RPC] (3) Storage Info @/ext SUCCESS
293348 [RPC] (4) Storage Stat @/ext/Manifest START
293432 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
293447 [RPC] (5) System Get DateTime START
293537 [DBG] Flipper time skew is -1654 milliseconds
293539 [RPC] (5) System Get DateTime SUCCESS
293548 [RPC] (6) System Set DateTime START
293617 [RPC] (6) System Set DateTime SUCCESS
293617 [RPC] Stopping RPC session...
293623 [RPC] Attempting to unload protobuf plugin...
293623 [RPC] Unloaded protobuf plugin.
293623 [RPC] RPC session stopped successfully.
293623 [REG] Registering the device
293623 [DEV] Version: 0.93.9 commit: a815d93e radio: 1.17.3
293623 [BKD] Current device changed to "Kindle"
293729 [RPC] Starting RPC session...
294465 [RPC] RPC session started successfully.
294467 [RPC] (1) Gui Start ScreenStream START
294568 [RPC] (1) Gui Start ScreenStream SUCCESS
378902 [RPC] Serial connection was lost.
378902 [RPC] Stopping RPC session...
378903 [RPC] Attempting to unload protobuf plugin...
378904 [RPC] Unloaded protobuf plugin.
378904 [RPC] RPC session stopped successfully.
600406 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
600478 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
824095 [REG] Device disconnected: VID_0x483:PID_0x5740
824095 [BKD] Last device was disconnected
839298 [REG] Detected new device: VID_0x483:PID_0x5740
839307 [DBG] Trying serial port FLIP_R0WTUMZ at \\.\COM5
839307 [DBG] Using  serial port FLIP_R0WTUMZ at \\.\COM5
839307 [RPC] Starting RPC session...
839345 [RPC] RPC session started successfully.
839345 [RPC] (1) System Protobuf Version START
839409 [DBG] Detected protobuf version: 0.20
839409 [RPC] (1) System Protobuf Version SUCCESS
839409 [RPC] (2) Property Get START
839552 [RPC] (2) Property Get SUCCESS
839552 [RPC] (3) Storage Info @/ext START
839616 [RPC] (3) Storage Info @/ext SUCCESS
839616 [RPC] (4) Storage Stat @/ext/Manifest START
839669 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
839672 [RPC] (5) System Get DateTime START
839711 [DBG] Flipper time skew is -1828 milliseconds
839711 [RPC] (5) System Get DateTime SUCCESS
839711 [RPC] (6) System Set DateTime START
839774 [RPC] (6) System Set DateTime SUCCESS
839774 [RPC] Stopping RPC session...
839791 [RPC] Attempting to unload protobuf plugin...
839791 [RPC] Unloaded protobuf plugin.
839791 [RPC] RPC session stopped successfully.
839792 [REG] Registering the device
839792 [DEV] Version: 0.93.9 commit: a815d93e radio: 1.17.3
839792 [BKD] Current device changed to "Kindle"
839901 [RPC] Starting RPC session...
839948 [RPC] RPC session started successfully.
839948 [RPC] (1) Gui Start ScreenStream START
840017 [RPC] (1) Gui Start ScreenStream SUCCESS
842532 [RPC] Serial connection was lost.
842532 [RPC] Stopping RPC session...
842532 [RPC] Attempting to unload protobuf plugin...
842532 [RPC] Unloaded protobuf plugin.
842532 [RPC] RPC session stopped successfully.
```

### Anything else?

Other than this, great work folks as always thank you

RogueMaster, 2023-12-14:
Are you using the latest No Animations Build?

aoshea83, 2023-12-14:
Hi

Yest I've just updated to RM1207-2110-0.93.9-a815d93-NoAnim.zip

Kind regards,
___________________

Alexander O' Shea
+353 (87) 251 0720

RogueMaster, 2023-12-14:
This should be fixed for next build

closed 2023-12-14 by RogueMaster