RogueMaster/flipperzero-firmware-wPlugins #1008: WiFi Marauder companion in github version does not save data to pcap

LowSkillDeveloper, 2023-12-14:
### Describe the bug.

I have the latest version for github installed and in it WiFi Marauder companion does not save pcap files correctly, all files are 0 bytes in size. 
At the beginning, I thought the problem was in the firmware for the wifi dev board, but flashing different versions did not help. 
Then I downloaded the latest version of companion from [here](https://github.com/0xchocolate/flipperzero-wifi-marauder/actions/workflows/build.yml), put it in the gpio folder and launched it. And it saves pcap files normally, although at the beginning it writes a warning about mismatch api.

### Reproduction

1. Open WiFi Marauder companion
2. Enable save pcap on flipper sdcard
3. Sniff
4. Open the folder and see that all files are empty, 0 bytes each

RogueMaster, 2023-12-14:
You likely don't have the right marauder installed on your ESP32. Please retry RM version if you got the other working.

closed 2023-12-14 by RogueMaster

LowSkillDeveloper, 2023-12-14:
Can you tell me where to find it? In the repository description there is a link to the old version from tcpassos, https://github.com/tcpassos/ESP32Marauder.

But i always installed the version from the justcallmekoko repository, https://github.com/justcallmekoko/ESP32Marauder

I have this version installed https://github.com/justcallmekoko/ESP32Marauder/releases/tag/v0.13.5
I flash esp32_marauder_v0_13_5_20231126_**flipper_sd_serial**.bin

I just re-flashed my ESP32 and try again but WiFi Marauder which is located in the GPIO/ESP32 folder, saves empty pcap files. And the version that I downloaded and drop just in GPIO folder [flipperzero-wifi-marauder-f7-api-49.1-unlshd-066](https://github.com/0xchocolate/flipperzero-wifi-marauder/actions/runs/7203905020), it saves data in pcap.

RogueMaster, 2023-12-14:
Thank you, will review and check into and fix.