RogueMaster/flipperzero-firmware-wPlugins #1012: nfc emulation doesn't seem to work whenever i install

gizmogamesyt, 2023-12-20:
### Describe the bug.

i'm doing everything propterly, or so i think, but i can't get the nfc emulation at all to work

### Reproduction

it happens during install

RogueMaster, 2023-12-20:
Please explain exactly what build you are using and everything you did to make you think you were doing it "properly"

closed 2023-12-20 by RogueMaster

gizmogamesyt, 2023-12-20:
i used the web installer

gizmogamesyt, 2023-12-20:
> Please explain exactly what build you are using and everything you did to make you think you were doing it "properly"

i used the web installer and it says it can't parse the uid for amiibos and other nfc that i've saved

gizmogamesyt, 2023-12-20:
https://github.com/RogueMaster/flipperzero-firmware-wPlugins/issues/1012#issuecomment-1864145841_