RogueMaster/flipperzero-firmware-wPlugins #1014: Evil Portal doesn't broadcast an AP/SSID

jonnyborbs, 2023-12-21:
### Describe the bug.

I've been trying for days to get this working but the Evil Portal FAP (or the Evil Portal option/flow within the Marauder FAP) claim they are working properly, but don't actually do anything. Configuring the official dev board with the 0.13.5 or 0.13.6 firmware, in both SD and non-SD card mode (I personally have a v1 dev board modded with the JustCallMeKoko SD daughterboard) makes no difference. 

Symptoms: 
- The marauder functionality works properly and packet captures/AP sniffing works but starting the evil portal reads in the ap.config.txt file  as well as the index.html file (either from the devboard internal SD card or the Flipper SD card depending on which firmware was loaded)
- Starting the portal claims to have run with the correct AP name and index.html file, but the devboard LED remains blue, not green as expected
- No SSID is visible on broadcast
- Ending the portal with stopscan turns up a truncated log entry that I can't get all of (not sure why it truncates) along the lines of "could not deinit wifi driver"

### Reproduction

I only have my own hardware so here's what I do:

Flipper Zero FZ.1
Official Wifi Dev Board v1 ESP32 (with SD card daughterboard)
Firmware: RM1215-1347-0.93.9-073ec96
Wifi Firmware: ESP32 Marauder 0.13.6 esp32_marauder_v0_13_6_20231214_flipper.bin or 0.13.5 as included with the ESP flasher in the above version of RM firmware, both produce the same results

Configure an ap.config.txt with the AP name
Create an index.html (I am using the Apple one from the Evil Portals repo because it is very small - 9kb - and wanted to reduce the chance a large portal size was causing issues)

Open ESP32 Marauder and select Start Evil Portal
Notice the LED is blue, not green, there is no AP/SSID broadcast, and. that sending stopscan produces an error

RogueMaster, 2023-12-21:
Please make sure you have the latest Marauder installed on your device

closed 2023-12-21 by RogueMaster

jonnyborbs, 2023-12-21:
I definitely do - latest firmware was specified above. 0.13.6 built like two days ago.