RogueMaster/flipperzero-firmware-wPlugins #1026: Amiibo Remixer (Current Bounty: 3 Months Subscriber Builds)

RogueMaster, 2024-01-29:
### Description of the feature you're suggesting.

An app that would emulate an existing NFC Amiibo from SD. However, randomize the UID and re-sign, and allow option to emulate remixed and save remixed NFC Amiibo.

### Anything else?

This allows for quick Zelda loot farming and facing off 5 of the same Lvl 50 Amiibo in Smash.

RogueMaster, 2024-04-28:
@GMMan or @noproto any interest? I'll double the bounty ...

RogueMaster, 2024-04-28:
Or @Willy-JL?

kcoderhtml, 2024-08-19:
I would be interested in this! Is the bounty planning on changing anytime soon?

RogueMaster, 2024-08-19:
> I would be interested in this! Is the bounty planning on changing anytime soon?

No, bounty remains. 😄

bettse, 2025-04-23:
Related: https://github.com/bettse/weebo

RogueMaster, 2025-04-23:
Thank you @bettse ! BTC sent 😄 🚀

closed 2025-04-23 by RogueMaster