RogueMaster/flipperzero-firmware-wPlugins #1030: FBT ERRORS

cokyrain, 2024-02-03:
### Describe the bug.

	CC	applications/external/totp/cli/plugins/timezone/timezone.c
applications/main/nfc/plugins/supported_cards/emv.c: In function 'emv_parse':
applications/main/nfc/plugins/supported_cards/emv.c:76:15: error: 'EmvApplication' has no member named 'name_found'
   76 |         if(app.name_found)
      |               ^
scons: *** [build/f7-firmware-C/.extapps/emv_parser/plugins/supported_cards/emv.o] Error 1

********** FBT ERRORS **********
build/f7-firmware-C/.extapps/emv_parser/plugins/supported_cards/emv.o: Error 1
Error: Process completed with exit code 2.

### Reproduction

Branch: refactor

RogueMaster, 2024-02-04:
There are no compile issues, try ./fbt updater_package -c to clear your cache and try again

RogueMaster, 2024-02-04:
Looking into this further as it does seem to be an issue...

RogueMaster, 2024-02-04:
Fixed

closed 2024-02-04 by RogueMaster