RogueMaster/flipperzero-firmware-wPlugins #1031: ISO 15693 SLIX2 - Can not be saved - AM.PASS

xsoft, 2024-02-06:
### Describe the bug.

**Card type**: ISO 15693 SLIX2
**Specifically**: AM.PASS (Pump It Up card, from Andamiro)

can be read,
can be emulated
can **NOT** be saved - error "_Can not save key file_" is shown.

Note that this can be done at OFW, even the older one.
But not at Rogue (tested on 0.93.9 31-01-2024)

### Reproduction

1. NFC
2. read ( "ISO 15693 SLIX2" card)
3. save > and you get error

### Target

ISO 15693 SLIX2

### Logs

```Text
This is from OFW flipper save (note that Ive censored priv./uniq. data with "xx", put any HEX there)
-------------------------------------------------------
Filetype: Flipper NFC device
Version: 4
# Device type can be ISO14443-3A, ISO14443-3B, ISO14443-4A, ISO14443-4B, ISO15693-3, FeliCa, NTAG/Ultralight, Mifare Classic, Mifare DESFire, SLIX, ST25TB
Device type: SLIX
# UID is common for all formats
UID: E0 04 01 xx xx xx xx E2
# ISO15693-3 specific data
# Data Storage Format Identifier
DSFID: 00
# Application Family Identifier
AFI: 00
# IC Reference - Vendor specific meaning
IC Reference: 01
# Lock Bits
Lock DSFID: false
Lock AFI: false
# Number of memory blocks, valid range = 1..256
Block Count: 80
# Size of a single memory block, valid range = 01...20 (hex)
Block Size: 04
Data Content: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
# Block Security Status: 01 = locked, 00 = not locked
Security Status: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
# SLIX specific data
# Passwords are optional. If a password is omitted, a default value will be used
Password Read: 00 00 00 00
Password Write: 00 00 00 00
Password Privacy: FF FF FF FF
Password Destroy: FF FF FF FF
Password EAS: 00 00 00 00
# This is the card's secp128r1 elliptic curve signature. It can not be calculated without knowing NXP's private key.
Signature: CD xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx F0
Privacy Mode: false
# Protection pointer configuration
Protection Pointer: 00
Protection Condition: 00
# SLIX Lock Bits
Lock EAS: false
Lock PPL: false
-------------------------------------------------------
```

### Anything else?

 "_Can not save key file_" is shown sometimes like on bank card to avoid confusion (according Google). I guess that here its misconfigured.

manspace80, 2024-02-06:
The problem with "Can not save key file" also occurs with Unlock SLIX-L

Reproduction:
1. NFC
2. Extra Actions
3. Unlock SLIX-L
4. Auth As TonieBox

After scanning a figure the error occurs

RogueMaster, 2024-02-06:
We will be getting rid of 0.93.9 to be replaced with refactors branch.

closed 2024-02-06 by RogueMaster

xsoft, 2024-02-06:
> We will be getting rid of 0.93.9 to be replaced with refactors branch.

Nice, thanks.
Would those file to be supported as well? (file from OFW cant be loaded)

```
Filetype: Flipper NFC device
Version: 4
Classic, Mifare DESFire, SLIX, ST25TB
Device type: SLIX
```

RogueMaster, 2024-02-06:
Someone on OFW team is working on it I think, if not, someone will have to develop it further on refactors

alexanderfilipzik, 2024-06-18:
Heyho I'm running into the same issue, is it already solved (RM0520-1640-0.102.1-13c0834)?