RogueMaster/flipperzero-firmware-wPlugins #1033: ESP32 Marauder throws "Cannot create app folder" after update to RM0208-2318-0.97.2-f563329

jonnyborbs, 2024-02-10:
### Describe the bug.

This just started after updating to RM0208-2318-0.97.2-f563329. 

I did delete the /apps folder before upgrading as I always do, particularly since this was a bump from 0.93 to 0.97

Now, when opening the ESP32 WiFi Marauder app it throws the following error
[image]

[image]

If you click past it, things seem to work OK still. The app is set to save all logs and PCAPs to the WiFi board's internal SD card. I can confirm I can list the contents of that SD card from within the app, and even did an update to Marauder 0.13.7 using the in-app flash utility - so it's not a hardware issue.

Not sure what this is about.

### Reproduction

- Update to RM0208-2318-0.97.2-f563329
- Open ESP32 WiFi Marauder
- Error is displayed

Can't confirm this is the case outside of my setup.

RogueMaster, 2024-02-10:
A marauder update will come soon, but I don't think it will fix this issue... we shall see...

jonnyborbs, 2024-02-10:
For what it's worth, I also checked out the ESP Flasher utility and noticed in this build, it has no app_data folder and thus no firmware binaries to swap between. IIRC that app and this one come from the same source, right?

RogueMaster, 2024-02-11:
Fixed by Latest Release RM0210-1603-0.97.2-2707a31

closed 2024-02-11 by RogueMaster