RogueMaster/flipperzero-firmware-wPlugins #1034: read transport card in France

KaliStudio, 2024-02-12:
### Description of the feature you're suggesting.

I would like to be able to read my French transport ticket and why not modify it, here are 4 dumps, the first three sound of the same title with 4, 3 and 2 trips remaining and the second there are 0 trips remaining.

[Liatick01022024_16h30.nfc.txt](https://github.com/DarkFlippers/unleashed-firmware/files/14233517/Liatick01022024_16h30.nfc.txt)
[Liatick10022024_18h20.nfc.txt](https://github.com/DarkFlippers/unleashed-firmware/files/14233518/Liatick10022024_18h20.nfc.txt)
[Liatick11022024_17h18.nfc.txt](https://github.com/DarkFlippers/unleashed-firmware/files/14233519/Liatick11022024_17h18.nfc.txt)
[Lia_10voy.nfc.txt](https://github.com/DarkFlippers/unleashed-firmware/files/14233520/Lia_10voy.nfc.txt)

RogueMaster, 2024-02-12:
Hi @KaliStudio , much of the work on NFC is being worked on by the OFW devs. Please open a ticket there.

closed 2024-02-12 by RogueMaster

KaliStudio, 2024-02-12:
The official developers don't want to work on it

RogueMaster, 2024-02-12:
Per my readme:

## This software is for experimental purposes only and is not meant for any illegal activity/purposes. We do not condone illegal activity and strongly encourage keeping transmissions to legal/valid uses allowed by law.

KaliStudio, 2024-02-12:
I would just like to retrieve the number of remaining trips, and display it but I don't know how to do it. The remaining number of trips corresponds to the first 2 digits of block 5

```
Block 0: 11 00 60 40
Block 1: 24 12 09 25
Block 2: 97 DA 9D 0C
Block 3: 00 00 4B F9
Block 4: 00 08 00 00
Block 5: 04 00 00 04 4 trips remaining
Block 6: F7 FF FF FF
Block 7: 18 18 1C B5
Block 8: 83 00 B7 42
Block 9: C1 45 05 C0
Block 10: 0F 25 00 D1
Block 11: 18 90 E0 B4
Block 12: 89 00 B7 42
Block 13: 09 46 08 C0
Block 14: 1F FC 00 51
Block 15: 54 B3 67 12
System OTP Block: FF FF FF FF
```