RogueMaster/flipperzero-firmware-wPlugins #1039: Web Installer type error

edmadrak, 2024-02-24:
### Describe the bug.

Clicked on the web installer button, when attempting to load the firmware it produced the following error:

TypeError: Cannot read properties of undefined (reading 'data')

Failed to fetch firmware: TypeError: Cannot read properties of undefined (reading 'data')
TypeError: Cannot read properties of undefined (reading 'data')

File Name used - "RM0222-1041-0.98.4-31a21ce"
URL Clicked - https://lab.flipper.net/?url=https%3A%2F%2Frogue-master.net%2F%3Ffile%3DRM0222-1041-0.98.4-31a21ce.tgz&channel=RM0222-1041-0.98.4-31a21ce&version=0.98.4

### Reproduction

1. Connect Flipper Zero and pair
2. Click above link
3. Click Install on RM firmware

gandlers, 2024-02-24:
Thought I was doing something wrong as I also had this issue.

RogueMaster, 2024-02-25:
Please use install with TGZ until it is resolved

edmadrak, 2024-02-27:
Confirmed works as expected. Many thanks for your continued efforts!