RogueMaster/flipperzero-firmware-wPlugins #1044: NFC program won't launch (crashes) on newest RogueMaster 0.99.1 firmware

Dochartaigh, 2024-03-09:
### Describe the bug.

NFC program fails to launch. Get a "Flipper crashed and was rebooted" "out of memory" error after the crash.
[image]
[image]

### Reproduction

1. From homescreen of RogueMaster (RM) press the center button.
2. Scroll down to "NFC" and press center button to launch it (*program crashes and shows error message posted above).

ALT: Can also press center button, then Apps, then NFC, then "NFC" program (which crashes as well).

### Logs

```Text
If you can post a tutorial/video on how to do these I would be glad to post the log...
```

### Anything else?

Just to note that I've reinstalled the newest 0.99.1 RogueMaster firmware no less than 3x times and the problem is the same ever time... Think I've been able to access the NFC program maybe 2 times... ALL the other times (dozens) it crashes.

RogueMaster, 2024-03-09:
This isn't happening, can't reproduce...

closed 2024-03-09 by RogueMaster

ahmadardal, 2024-03-12:
This is happening me as well..

RogueMaster, 2024-03-12:
I've heard there is a known issue with animations not getting removed from memory properly and causing NFC to crash and it is being looked into. This issue occurs on OFW as well.

Dochartaigh, 2024-03-12:
> I've heard there is a known issue with animations not getting removed from memory properly and causing NFC to crash and it is being looked into. This issue occurs on OFW as well.

On Roguemaster firmware I've fun out of fingers to count how many times the homescreen animation comes up missing (just blank, no animation/art, nothing there but battery/SDCard icons) when I haven't used NFC (or even RFID) App since it was booted-up... so on the animations itself I'm talking about, some code is buggy with those to begin with – outside of any NFC issues.

Back to this though, forget where I posted about this before, but I mentioned how my 4th? install of the newest RM firmware and NFC is working WAY better. Still gives me that error upon opening the program some rare times (when it immediately crashes/program doesn't open), but it's usable now at least. Will usually always launch, and if not only takes a time (maybe two) to open the NFC program. Before, it would crash 50 times in a row if I kept trying it... and I just couldn't get into NFC program no matter what - no matter how many times I rebooted the device (and even re-flashing RM FW at least 3x times - had the same issue on ALL of those installs for whatever reason).

RogueMaster, 2024-03-12:
> On Roguemaster firmware I've fun out of fingers to count how many times the homescreen animation comes up missing (just blank, no animation/art, nothing there but battery/SDCard icons) when I [...]

Seems you need to set your animation preference. In CFW Settings => Interface => Desktop => Animations, choose your animations build you want (not default). If you don't have options, get them [here](https://github.com/RogueMaster/awesome-flipperzero-withModules/tree/rogue_main/graphics/dolphin-ManifestSwitcher) and put those files into your dolphin folder (folders and txt files).

RogueMaster, 2024-03-12:
> Back to this though, forget where I posted about this before, but I mentioned how my 4th? install of the newest RM firmware and NFC is working WAY better. Still gives me that error upon opening the [...]

Once again, known issue being looked into