RogueMaster/flipperzero-firmware-wPlugins #1049: qflipper software does not see the device with rogue master installed

Hitmanreis, 2024-03-24:
### Describe the bug.

When rogue master is installed, qflipper software does not see the device. When I install the original flipper software on flipper zero, qflipper sees my device. How can I solve this problem?

### Reproduction

When I plug in the usb cable and open the qflipper app
[image]

RogueMaster, 2024-03-24:
Restart qflipper, restart flipper, make sure you are plugged in

closed 2024-03-24 by RogueMaster

Hitmanreis, 2024-03-24:
> Qflipper'ı yeniden başlatın, flipper'ı yeniden başlatın, fişinizin takılı olduğundan emin olun
I have tried everything, it works fine in recovery mode and on the web updater, I only get an error when qflipper is in normal mode, I have checked the usb driver and ports, there are no conflicts etc. There is no problem.

RogueMaster, 2024-03-24:
Indeed, there is no problem. Everything works as intended. If you are still having issues, try the Web Installer as that works without issues as well. Close qFlipper to use the Web Installer.