RogueMaster/flipperzero-firmware-wPlugins #1052: NFC Bug in (RM0329-0331-0.100.2-95748e8)

freesofia, 2024-03-29:
### Describe the bug.

After the last update, which does not go through the WEB Installer (only manually through the qFlipper app with file: [image].tgz), the device freezes and restarts when selecting NFC!

### Reproduction

1. Switch on
2. Press the button select app
3. Select NFC
4. Crash & Reboot

gades, 2024-03-29:
The same issue for me

RogueMaster, 2024-03-29:
This will be checked into but is a known issue being worked on

closed 2024-03-29 by RogueMaster