RogueMaster/flipperzero-firmware-wPlugins #1057: cc1101 device not detected

slikkmulah, 2024-04-17:
### Describe the bug.

newest version is not recognizing the cc1101 module for sub-ghz.

### Reproduction

237 [APP] qFlipper version 1.3.3 commit bfce851d 2023-11-10T01:40:31
237 [APP] OS info: Windows 10 Version 22H2 10 10.0.19045 Qt 6.4.2
414 [DBG] Trying serial port 00E04C239987 at \\.\COM10
414 [DBG] Trying serial port FLIP_GIZM0 at \\.\COM5
414 [DBG] Using  serial port FLIP_GIZM0 at \\.\COM5
414 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
414 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
419 [RPC] Starting RPC session...
480 [RPC] RPC session started successfully.
480 [RPC] (1) System Protobuf Version START
548 [DBG] Detected protobuf version: 0.22
548 [RPC] (1) System Protobuf Version SUCCESS
548 [RPC] (2) Property Get START
696 [RPC] (2) Property Get SUCCESS
696 [RPC] (3) Storage Info @/ext START
748 [RPC] (3) Storage Info @/ext SUCCESS
748 [RPC] (4) Storage Stat @/ext/Manifest START
812 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
812 [RPC] (5) System Get DateTime START
864 [DBG] Flipper time skew is -4090 milliseconds
864 [RPC] (5) System Get DateTime SUCCESS
864 [RPC] (6) System Set DateTime START
916 [RPC] (6) System Set DateTime SUCCESS
917 [RPC] Stopping RPC session...
922 [RPC] Attempting to unload protobuf plugin...
922 [RPC] Unloaded protobuf plugin.
922 [RPC] RPC session stopped successfully.
922 [REG] Registering the device
922 [DEV] Version: 0.100.4 commit: 22e2176d radio: 1.19.0
922 [BKD] Current device changed to "GIZM0"
1030 [RPC] Starting RPC session...
1096 [RPC] RPC session started successfully.
1096 [RPC] (1) Gui Start ScreenStream START
1163 [RPC] (1) Gui Start ScreenStream SUCCESS
5095 [RPC] (2) Gui Send Input START
5148 [RPC] (2) Gui Send Input SUCCESS
5213 [RPC] (3) Gui Send Input START
5265 [RPC] (3) Gui Send Input SUCCESS
5265 [RPC] (4) Gui Send Input START
5319 [RPC] (4) Gui Send Input SUCCESS
6562 [RPC] (5) Gui Send Input START
6614 [RPC] (5) Gui Send Input SUCCESS
6679 [RPC] (6) Gui Send Input START
6731 [RPC] (6) Gui Send Input SUCCESS
6731 [RPC] (7) Gui Send Input START
6796 [RPC] (7) Gui Send Input SUCCESS
10662 [RPC] (8) Gui Send Input START
10716 [RPC] (8) Gui Send Input SUCCESS
10764 [RPC] (9) Gui Send Input START
10828 [RPC] (9) Gui Send Input SUCCESS
10828 [RPC] (10) Gui Send Input START
10881 [RPC] (10) Gui Send Input SUCCESS
10982 [RPC] (11) Gui Send Input START
11044 [RPC] (11) Gui Send Input SUCCESS
11097 [RPC] (12) Gui Send Input START
11149 [RPC] (12) Gui Send Input SUCCESS
11149 [RPC] (13) Gui Send Input START
11213 [RPC] (13) Gui Send Input SUCCESS
11245 [RPC] (14) Gui Send Input START
11297 [RPC] (14) Gui Send Input SUCCESS
11395 [RPC] (15) Gui Send Input START
11448 [RPC] (15) Gui Send Input SUCCESS
11448 [RPC] (16) Gui Send Input START
11512 [RPC] (16) Gui Send Input SUCCESS
11619 [RPC] (17) Gui Send Input START
11678 [RPC] (17) Gui Send Input SUCCESS
11779 [RPC] (18) Gui Send Input START
11830 [RPC] (18) Gui Send Input SUCCESS
11830 [RPC] (19) Gui Send Input START
11896 [RPC] (19) Gui Send Input SUCCESS
12479 [RPC] (20) Gui Send Input START
12530 [RPC] (20) Gui Send Input SUCCESS
12614 [RPC] (21) Gui Send Input START
12666 [RPC] (21) Gui Send Input SUCCESS
12666 [RPC] (22) Gui Send Input START
12729 [RPC] (22) Gui Send Input SUCCESS
13828 [RPC] (23) Gui Send Input START
13880 [RPC] (23) Gui Send Input SUCCESS
13931 [RPC] (24) Gui Send Input START
13984 [RPC] (24) Gui Send Input SUCCESS
13984 [RPC] (25) Gui Send Input START
14046 [RPC] (25) Gui Send Input SUCCESS
15812 [RPC] (26) Gui Send Input START
15864 [RPC] (26) Gui Send Input SUCCESS
15961 [RPC] (27) Gui Send Input START
16014 [RPC] (27) Gui Send Input SUCCESS
16014 [RPC] (28) Gui Send Input START
16093 [RPC] (28) Gui Send Input SUCCESS

### Target

module - internal

### Logs

```Text
237 [APP] qFlipper version 1.3.3 commit bfce851d 2023-11-10T01:40:31
237 [APP] OS info: Windows 10 Version 22H2 10 10.0.19045 Qt 6.4.2
414 [DBG] Trying serial port 00E04C239987 at \\.\COM10
414 [DBG] Trying serial port FLIP_GIZM0 at \\.\COM5
414 [DBG] Using  serial port FLIP_GIZM0 at \\.\COM5
414 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
414 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
419 [RPC] Starting RPC session...
480 [RPC] RPC session started successfully.
480 [RPC] (1) System Protobuf Version START
548 [DBG] Detected protobuf version: 0.22
548 [RPC] (1) System Protobuf Version SUCCESS
548 [RPC] (2) Property Get START
696 [RPC] (2) Property Get SUCCESS
696 [RPC] (3) Storage Info @/ext START
748 [RPC] (3) Storage Info @/ext SUCCESS
748 [RPC] (4) Storage Stat @/ext/Manifest START
812 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
812 [RPC] (5) System Get DateTime START
864 [DBG] Flipper time skew is -4090 milliseconds
864 [RPC] (5) System Get DateTime SUCCESS
864 [RPC] (6) System Set DateTime START
916 [RPC] (6) System Set DateTime SUCCESS
917 [RPC] Stopping RPC session...
922 [RPC] Attempting to unload protobuf plugin...
922 [RPC] Unloaded protobuf plugin.
922 [RPC] RPC session stopped successfully.
922 [REG] Registering the device
922 [DEV] Version: 0.100.4 commit: 22e2176d radio: 1.19.0
922 [BKD] Current device changed to "GIZM0"
1030 [RPC] Starting RPC session...
1096 [RPC] RPC session started successfully.
1096 [RPC] (1) Gui Start ScreenStream START
1163 [RPC] (1) Gui Start ScreenStream SUCCESS
5095 [RPC] (2) Gui Send Input START
5148 [RPC] (2) Gui Send Input SUCCESS
5213 [RPC] (3) Gui Send Input START
5265 [RPC] (3) Gui Send Input SUCCESS
5265 [RPC] (4) Gui Send Input START
5319 [RPC] (4) Gui Send Input SUCCESS
6562 [RPC] (5) Gui Send Input START
6614 [RPC] (5) Gui Send Input SUCCESS
6679 [RPC] (6) Gui Send Input START
6731 [RPC] (6) Gui Send Input SUCCESS
6731 [RPC] (7) Gui Send Input START
6796 [RPC] (7) Gui Send Input SUCCESS
10662 [RPC] (8) Gui Send Input START
10716 [RPC] (8) Gui Send Input SUCCESS
10764 [RPC] (9) Gui Send Input START
10828 [RPC] (9) Gui Send Input SUCCESS
10828 [RPC] (10) Gui Send Input START
10881 [RPC] (10) Gui Send Input SUCCESS
10982 [RPC] (11) Gui Send Input START
11044 [RPC] (11) Gui Send Input SUCCESS
11097 [RPC] (12) Gui Send Input START
11149 [RPC] (12) Gui Send Input SUCCESS
11149 [RPC] (13) Gui Send Input START
11213 [RPC] (13) Gui Send Input SUCCESS
11245 [RPC] (14) Gui Send Input START
11297 [RPC] (14) Gui Send Input SUCCESS
11395 [RPC] (15) Gui Send Input START
11448 [RPC] (15) Gui Send Input SUCCESS
11448 [RPC] (16) Gui Send Input START
11512 [RPC] (16) Gui Send Input SUCCESS
11619 [RPC] (17) Gui Send Input START
11678 [RPC] (17) Gui Send Input SUCCESS
11779 [RPC] (18) Gui Send Input START
11830 [RPC] (18) Gui Send Input SUCCESS
11830 [RPC] (19) Gui Send Input START
11896 [RPC] (19) Gui Send Input SUCCESS
12479 [RPC] (20) Gui Send Input START
12530 [RPC] (20) Gui Send Input SUCCESS
12614 [RPC] (21) Gui Send Input START
12666 [RPC] (21) Gui Send Input SUCCESS
12666 [RPC] (22) Gui Send Input START
12729 [RPC] (22) Gui Send Input SUCCESS
13828 [RPC] (23) Gui Send Input START
13880 [RPC] (23) Gui Send Input SUCCESS
13931 [RPC] (24) Gui Send Input START
13984 [RPC] (24) Gui Send Input SUCCESS
13984 [RPC] (25) Gui Send Input START
14046 [RPC] (25) Gui Send Input SUCCESS
15812 [RPC] (26) Gui Send Input START
15864 [RPC] (26) Gui Send Input SUCCESS
15961 [RPC] (27) Gui Send Input START
16014 [RPC] (27) Gui Send Input SUCCESS
16014 [RPC] (28) Gui Send Input START
16093 [RPC] (28) Gui Send Input SUCCESS
```

RogueMaster, 2024-04-17:
Make sure CFW Settings => Protocols has values set to Default

slikkmulah, 2024-04-17:
> Make sure CFW Settings => Protocols has values set to Default

I checked that in the cfw settings
 and same issue (pic attached). I also tried a couple other cc1101 boards and when I try different FW or an older version it seems to work fine.

[image]

RogueMaster, 2024-04-18:
Please try this... 
[RM0418-0000-SubGRevert.zip](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/15019691/RM0418-0000-SubGRevert.zip)

If that works I will merge into main as a temporary fix...

slikkmulah, 2024-04-18:
> Please try this... [RM0418-0000-SubGRevert.zip](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/15019691/RM0418-0000-SubGRevert.zip)
> 
> If that works I will merge into main as a temporary fix...

That seems to have done the trick. :) works perfectly now.

RogueMaster, 2024-04-18:
Will be on next release

closed 2024-04-18 by RogueMaster

FatherDivine, 2024-04-20:
> Please try this... [RM0418-0000-SubGRevert.zip](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/files/15019691/RM0418-0000-SubGRevert.zip)
> 
> If that works I will merge into main as a temporary fix...

Any tips to installing with qflipper on linux? Every time I try to flash, it puts it in the DFU/recovery mode, then doesn't read so I have to restart qflipper app and reconnect flipper.. it recognizes then but still won't actually flash, as it gives errors.

40208 [DBG] Failed to read DfuSe file prefix
40208 [DBG] DfuSe file is not valid
40208 [RCY] Firmware Download @Ardolike ERROR: Can't flash firmware: An error has occurred during the operation.
40209 [DEV] Firmware install from file @Ardolike ERROR: Can't flash firmware: An error has occurred during the operation.
40209 [BKD] Current operation finished with error: "Can't flash firmware: An error has occurred during the operation."

RogueMaster, 2024-04-20:
The normal build should work fine, this is build into latest release.... you should extract folder to update folder and update via flipper

FatherDivine, 2024-04-20:
> The normal build should work fine, this is build into latest release.... you should extract folder to update folder and update via flipper

I tried reverting to 0416 didn't work. Then tried flashing the firmware.DFU file on top you attached, no luck. Tried the RM0417 as well on GH, no dice. Will wait til the next or next next release to try again. 

I can't use Mayhem V2 or LE Flux Capacitor cc1101's... works fine with ESP WIFI Marauder app though. Sub-GHz app is the only issue. It sends power to 5V when opening for a brief second then doesn't connect or show external as available. I have an XXXtender on my flipper so I can easily see when 5v is applied (there's a LED). Can apply 5v thru menu, but turns off when opening Sub-GHZ app.

FoxLy1312, 2024-04-21:
Had same problem, back on 0329 everything working