RogueMaster/flipperzero-firmware-wPlugins #1059: Sub-ghz

bindix3737, 2024-04-18:
### Description of the feature you're suggesting.

flipper zero frequency is out side of default range check dock.

India region problem

Not emulate the sub-ghz frequency

RogueMaster, 2024-04-18:
Did you try to bypass or extend ranges?

closed 2024-04-18 by RogueMaster

bindix3737, 2024-04-18:
Bank card emulate future required
Nfc Payment not emulated 
Any credit card or Debit card

bindix3737, 2024-04-18:
> Did you try to bypass or extend ranges?

Not work

RogueMaster, 2024-04-18:
> Bank card emulate future required Nfc Payment not emulated Any credit card or Debit card

None of these are supposed to work....

bindix3737, 2024-04-18:
> > Bank card emulate future required Nfc Payment not emulated Any credit card or Debit card
> 
> None of these are supposed to work....

Hmm...
Any reason

RogueMaster, 2024-04-18:
> > > Bank card emulate future required Nfc Payment not emulated Any credit card or Debit card
> > 
> > 
> > None of these are supposed to work....
> 
> Hmm... Any reason

You haven't made the functionality yet, seems like you want it... start learning and you may find its harder to get that feature to work than you think... but maybe you can figure it out and bring it to flipper for everyone...

bindix3737, 2024-04-18:
> You haven't made the functionality yet, seems like you want it... start learning and you may find its harder to get that feature to work than you think... but maybe you can figure it out and bring [...]

I don't know how to solve problems, so I'm telling you

bindix3737, 2024-04-18:
OK thanks,
But , I don't know how to solve problems, so I'm telling you