RogueMaster/flipperzero-firmware-wPlugins #1062: Keeloq decrypt?

majki09, 2024-04-22:
### Describe the enhancement you're suggesting.

Hi. I'm working on [Python script](https://github.com/majki09/flipperzero-tools/blob/main/keeloq.py) that manages all saved _.sub_ files from Keeloq keyfobs. So far, having a _key_ I can extract it's serial number and button that was used. I'm having problems with decrypted part of the _key_. Can anyone help me with this? I've seen the [keeloq_mfcodes](https://github.com/flipperdevices/flipperzero-firmware/blob/dev/applications/main/subghz/resources/subghz/assets/keeloq_mfcodes) file, but it seems to be also encrypted and I don't know how to use it.
Anyone? Thanks 😁

RogueMaster, 2024-04-22:
Sorry this isn't the right spot for this, you are welcomed to ask on our Discord...

closed 2024-04-22 by RogueMaster

donglegoodtime, 2024-04-25:
CUT yrK45