RogueMaster/flipperzero-firmware-wPlugins #1065: crashes when started

earlpfau, 2024-05-03:
### Describe the bug.

Hi no matter what happens if i restart my flipper it will crash if i push any other button other then the center button and this will happen every time until i push the center button.

### Reproduction

1. start up from off.
2. push any button other than the center button.
3. will continue to crash and restart till you push the center button.

### Target

unknown

RogueMaster, 2024-05-04:
Please delete the file at .config/cfw_settings.txt on SD card and let me know if it solves this issue.

RogueMaster, 2024-05-04:
Please do confirm if you have issues after this.

closed 2024-05-04 by RogueMaster

niggowai, 2024-05-09:
Had the same problem, deleting the file fixed it and the wallpapers also work again.

jonnyborbs, 2024-05-12:
I had tons of issues too (slowness and the crashing on any button press) - deleting this file solves it. But I can't change the screen animations now and I used to be able to. I actually prefer them disabled fully but the animations menu option only allows "Default"

RogueMaster, 2024-05-12:
Make sure you have the manifest_None.txt file