RogueMaster/flipperzero-firmware-wPlugins #1068: Sub-GHz app doesn't decode (Pocsag) anymore...

dl9sec, 2024-05-09:
### Describe the bug.

With the latest RM firmware the Sub-GHz app doesn't decode anymore (predecessor RM0429-1240-0.101.3-a5550b3 worked fine).
I can see and hear, that a signal is received, but there is no decoding (nor in "Read", neither in "Read RAW").

### Reproduction

1. Renamed "SD Card/subghz/assets/setting_user.txt" to "setting_user.txt.org".
2. Renamed "SD Card/subghz/assets/setting_user.pocsag" to "setting_user.txt".
3. Rebooted Flipper
4. Started "Sub-Ghz" -> "Read"
5. Configured frequency to "439.98" at the end of the frequency list

RogueMaster, 2024-05-10:
Use POCSAG app for pocsag

closed 2024-05-10 by RogueMaster

dl9sec, 2024-05-10:
Sure, but not for recording pocsag transmissions

RogueMaster, 2024-05-10:
Understood will get re-added for next build. Could you help me understand, what can you do with a recorded POCSAG transmission?

dl9sec, 2024-05-11:
E.g. to help xMasterX to debug the Pocsag pager app (see https://github.com/xMasterX/flipper-pager/issues/14).
The Sub-Ghz app is very helpful here and it is a pitty when the app can not decode Pocsag correctly...

RogueMaster, 2024-05-11:
Damn, you don't have a real use case? It's a pitty you don't have a real use for this 🤦🏼