RogueMaster/flipperzero-firmware-wPlugins #1069: Can’t change desktop animations in 0.101.3 (5/6/2024)

jonnyborbs, 2024-05-12:
### Describe the bug.

Going to CFW Settings > Interface > Common > Desktop, the Animations menu is locked at Default, it can’t be changed.

I can see all the animations in the /ext/dolphin folder, but really I just want to turn them off and can’t

### Reproduction

Turn on
Go to CFW Settings > Interface > Common > Desktop, 
Note that the Animations menu is locked at Default, it can’t be changed.

RogueMaster, 2024-05-12:
Make sure you have manifest_None.txt etc

RogueMaster, 2024-05-12:
Here is the folder to copy to your SD card: https://github.com/RogueMaster/awesome-flipperzero-withModules/tree/rogue_main/graphics/dolphin-ManifestSwitcher

jonnyborbs, 2024-05-12:
I need to bring that whole folder over? This was a new install on a fresh SD card and here is what's in my /ext/dolphin folder now (only one manifest.txt)
[image]

RogueMaster, 2024-05-12:
copy the [manifest_None.txt](https://github.com/RogueMaster/awesome-flipperzero-withModules/blob/rogue_main/graphics/dolphin-ManifestSwitcher/manifest_None.txt) and [manifest_Minimal.txt](https://github.com/RogueMaster/awesome-flipperzero-withModules/blob/rogue_main/graphics/dolphin-ManifestSwitcher/manifest_Minimal.txt) onto your SD. What build did you install?

jonnyborbs, 2024-05-13:
I installed [RM0506-1236-0.101.3-f0871a4](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/releases/tag/RM0506-1236-0.101.3-f0871a4) - I've been upgrading in place via qFlipper for nearly a year now, but this build wouldn't copy properly to the SD card so I wiped the entire device and installed it fresh via the web UI.

copying those files brought back the minimal and none options, so that's great - I just wonder if something was borked in this most recent build that caused all this info not to come over!

I did notice the file size was weird this time around. Normally I feel like it was ~5MB, this time it was much larger

jonnyborbs, 2024-05-13:
FWIW I also had the issue mentioned in https://github.com/RogueMaster/flipperzero-firmware-wPlugins/issues/1065 with this fresh build - every button press causing a crash, until I deleted a file. But that file would have been installed fresh with this new build.

RogueMaster, 2024-05-13:
Understood, yeah after some changes that file had to be deleted it seems 🤦🏼 

As for this issue, the dolphin folder and its manifest_whatever files don't ever get deleted from install... so if you wiped the dolphin folder, that would explain the lack of animation options, to get more options, just copy this folder https://github.com/RogueMaster/awesome-flipperzero-withModules/tree/rogue_main/graphics/dolphin-ManifestSwitcher to your SD card

closed 2024-05-13 by RogueMaster