RogueMaster/flipperzero-firmware-wPlugins #1072: USB Mouse Jiggler doesn't work anymore

dl9sec, 2024-05-19:
### Describe the bug.

When the Mouse Jiggler app is started the flipper isn't recognized as a HID mouse.
Nothing happens if the jiggler is set to "RUNNING...".

### Reproduction

- Start USB Mouse Jiggler app
- Plug in USB to PC
- Start Mouse Jiggler

RogueMaster, 2024-05-19:
Definitely works for me... if you continue to have issues with it, please PR your solution so others with the same issue can have a fix...

closed 2024-05-19 by RogueMaster

dl9sec, 2024-05-20:
Works perfect with RM0429-1240-0.101.3-a5550b3 but doesn't work anymore for any further version since now.
Installed the latest version from the scratch (formatted SD, flashed OFW, deleted /ext/apps, updated with the Webinstaller).
Normally as soon as I start the Mouse Jiggler, I can hear, that the USB detected a new device (in this case a HID mouse). With the latest version nothing changes on USB, no HID mouse.
Going back to RM0429-1240-0.101.3-a5550b3 Mouse Jiggler works fine...

dl9sec, 2024-05-23:
Since RM0520-1640-0.102.1-13c0834 the Mouse Jiggler works again...