RogueMaster/flipperzero-firmware-wPlugins #1082: Flipper Bricked after selecting sword icon from quick panel menu

FuckingToasters, 2024-07-29:
### Describe the bug.

So after I selected the swored icon, I got to a screen were I only see small as card icon and battery at the top. Pressing right or left arrow shoes different images but I can't get back to were I were before. I've tried rebooting but it didn't help.

### Reproduction

1. Click on sword icon in Quickmenu Panel
2. See battery icons
3. Reboot
4. Never see the good flipper stuff again (apps, files etc.)

FuckingToasters, 2024-07-29:
[image]

RogueMaster, 2024-07-29:
Press up to access menu and exit the sword mode, and please don't open an issue for not knowing how to use the flipper. You can search for answers on Discord.

closed 2024-07-29 by RogueMaster