RogueMaster/flipperzero-firmware-wPlugins #1093: Reboots

refnauj10, 2024-08-25:
### Describe the bug.

There is a moment when the flipper zero reboots at least 5 times before stabilizing, indicating that the furi load has failed

### Reproduction

1. start a function 
or plugin
2. start a menu feature
3. First start

RogueMaster, 2024-08-25:
Make sure you have selected an animation set in CFW Settings => Interface => Desktop => Animations

RogueMaster, 2024-08-25:
Since its having issues pick No Anim

closed 2024-08-25 by RogueMaster

refnauj10, 2024-08-25:
Well, an the moment this option is on “none” but the reboots persists…

Michael-A-Sloan, 2024-09-16:
This is still an Issue with mine as well on the latest firmware.

Kitt3120, 2024-09-22:
Still having the same issue on 0919-0056

huuck, 2025-02-13:
still happening.