RogueMaster/flipperzero-firmware-wPlugins #1104: Furi Check Failed

SneezyLlama, 2024-10-03:
### Describe the bug.

Whenever the flipper is powered off and then on again, it crashes and reboots with a message saying "furi_check failed".

### Reproduction

Switch on the flipper.

closed 2024-10-03 by RogueMaster

benleb, 2024-10-13:
This issue still exists (or came back) with the current 420 branch

RogueMaster, 2024-10-13:
Delete your dolphin and app folders and re-install the firmware 😄

benleb, 2024-10-13:
okay, thanks 😄👍

ATr0jan, 2024-10-24:
I Reflashed with DFU back to stock and reformatted the SD Card before reloading the Latest RM firmware and i am still seeing this Error

A2-Dev, 2024-12-20:
FWIW i tried installing RM multiple ways (copying files to SD, coping to update folder, qflipper) and kept getting this error even after i deleted apps and dolphin dirs and reinstalled. Would only crash on boot up if SD card was inserted. if you start it up without SD card inserted and then insert SD card it would work fine. Crash once then it worked fine if you started with SD card installed.

Tried today with a smaller SD card 32gb Samsung Evo instead of the original 64gb Samsung Evo). Not sure if someone else could try and reproduce. Just curious if it's the SD card as I've seen it cause RPi issues in the past.

RogueMaster, 2024-12-20:
It's very likely an SD thing since others have had the error subside after swapping SD

huuck, 2025-02-13:
It's not an SD thing, it's still happening even after testing 3 different SD cards.