RogueMaster/flipperzero-firmware-wPlugins #1105: ram/ Memory problems

helix19880, 2024-10-14:
### Description of the feature you're suggesting.

Sort of need a programming so I'm more or less just gonna give you the code and say here this is my idea. Hopefully we can implement it to solve some of these problems. NFC Reader Constant. creates an available memory problem. So I wrote a simple. ram cleaner slash. cache cleaner. And more or less just needs to be refined a little bit. and targeted. Hopefully this at least lights up a light bulb. and gets us in the right direction for no longer having memory issues. I'm sure there's some way where this could be automated in the background. If anything, just to create a better quality of life while playing with the flipper zero And I do believe this will be a new function.

#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <flipper.h>

void clean_ram() {
    // Simulate RAM cleaning by freeing unused memory
    printf("Cleaning RAM...\n");
    fflush(stdout);
    // Perform garbage collection or memory optimization here
    // Placeholder for actual memory management code
    printf("RAM cleaned successfully!\n");
}

void clear_cache() {
    // Simulate clearing cache files
    printf("Clearing cache files...\n");
    fflush(stdout);
    // Placeholder for actual cache clearing code
    // You can add specific functions to delete cache files as needed
    printf("Cache files cleared successfully!\n");
}

int main(void) {
    printf("Starting RAM cleaner and cache clearer...\n");
    clean_ram();
    clear_cache();
    printf("RAM cleaner and cache clearer finished.\n");
    return 0;
}

### Anything else?

#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <flipper.h>

void clean_ram() {
    // Simulate RAM cleaning by freeing unused memory
    printf("Cleaning RAM...\n");
    fflush(stdout);
    // Perform garbage collection or memory optimization here
    // Placeholder for actual memory management code
    printf("RAM cleaned successfully!\n");
}

void clear_cache() {
    // Simulate clearing cache files
    printf("Clearing cache files...\n");
    fflush(stdout);
    // Placeholder for actual cache clearing code
    // You can add specific functions to delete cache files as needed
    printf("Cache files cleared successfully!\n");
}

int main(void) {
    printf("Starting RAM cleaner and cache clearer...\n");
    clean_ram();
    clear_cache();
    printf("RAM cleaner and cache clearer finished.\n");
    return 0;
}

RogueMaster, 2024-10-14:
Enlightening 😆 If only it were this simple

closed 2024-10-14 by RogueMaster

RogueMaster, 2024-11-10:
It should run out of memory less now

sfjuocekr, 2024-11-18:
> It should run out of memory much less now

You can quite reliably get the NFC app to crash:

NFC > Read > More > Info > More > (crash)

I must say the above pseudo code is top notch, if only it was that simple and all memory problems would be a thing of the past!

RogueMaster, 2025-04-15:
See if latest changes have fixed your issue

sfjuocekr, 2025-04-21:
Will test tomorrow!