RogueMaster/flipperzero-firmware-wPlugins #1122: NiceFlo cant decoded in Sub-GHZ

Penelopen, 2025-01-03:
### Describe the bug.

Nice Flo 12/20bit or Nice FloR 56bit are not decoded in flipper. It not know about these systems, but I can create this remotes manually..

Even if I manually create these remotes in flipper, that my another flipper cant see that.

Flipper can create Nice Flo/FloR but can't decode it.
Reproduction

Open Sub-GHZ Read
Target

Sub-GHZ

### Reproduction

As in description

closed 2025-01-04 by RogueMaster

Penelopen, 2025-01-13:
I not see that fix. Why it closed?

RogueMaster, 2025-01-13:
> I not see that fix. Why it closed?

Our NiceFlo support is the best available amongst CFW, since the feature is the same as other CFW. However, we don't have support to expand upon the current implementation and we did not create it. So therefore, this is not the place for this ticket.