RogueMaster/flipperzero-firmware-wPlugins #1124: bus fault when write nfc tag

aurelx03, 2025-01-06:
### Describe the bug.

hi,

after a writing of nfc tag "miflare classic 1k" flipper reboot with error "bus fault"
same with a new saved of tag
worked before without error

writing seem to be effective regardless of that but its annoying

### Reproduction

switch on
enter
start nfc
saved
choose a saved file
writing on initial card
crash and reboot with bus fault

### Target

flipper nfc app

RogueMaster, 2025-01-08:
Try disabling BT and setting animations set to None

closed 2025-01-08 by RogueMaster

aurelx03, 2025-01-12:
same and same with RM0108-2246-0.420.0-6b6e4e8

RogueMaster, 2025-04-15:
See if latest changes have fixed your issue

aurelx03, 2025-04-15:
same with the last build, but issue same with other firmware, write correctly but reboot with bus fault