RogueMaster/flipperzero-firmware-wPlugins #1126: flipper crash duri_check failed

VERSE187, 2025-02-01:
### Describe the bug.

When I change the CFW settings and attempt to update the passport image, the flipper crashes every time, even after multiple reboots and installations on SD cards. The same issue persists. Additionally, when I try to run app that requires RAM, I receive an "out of RAM" error.

### Reproduction

1. Settings
2. interface
3. passport
4. image, change image and backout or reboot crashes all the time

### Logs

```Text

```

RogueMaster, 2025-02-01:
I tested with BT on and off and could not replicate. Try turning off BT and see if you still have issues. Also make sure you don't have extra files in your internal partition.

closed 2025-02-01 by RogueMaster

VERSE187, 2025-02-01:
It still happens despite wiping everything, even after factory reset. Bluetooth is off.