RogueMaster/flipperzero-firmware-wPlugins #1130: furi_check failed

LaJaiba, 2025-02-10:
### Describe the bug.

Every time Flipper turns on it crashes and then the message is "Flipper crashed and was rebooted furi_check failed" since the previous version this is constantly happening.

### Reproduction

1. Switch on.

### Logs

```Text

```

closed 2025-02-10 by RogueMaster

RogueMaster, 2025-02-13:
You should try removing apps folder and re-installing RM. This issue does not occur on release, along with the others.

Unc0v3RGD, 2025-06-08:
Same to me Hunduster, my flipper zero crashes 8 to 10 times everytime i make a change in cfw settings, taking 2 minutes to delete the file, i hope this gets solved soon, keep going with the great job @RogueMaster the custom firmware you developed is awesome, this had became my main flipper zero firmware since 2 months ago as i noticed it is extremely new in everything at cost of instability, really nice though