RogueMaster/flipperzero-firmware-wPlugins #1136: When attempting to read NFC keys, it restarts.

DroWez, 2025-04-13:
### Describe the bug.

When attempting to read NFC keys, it restarts. The logs do not show any entries, and the same restart occurs during manual key identification. A video is attached for reference. Video link [https://youtube.com/shorts/IXnaBLU_fHQ?feature=share](https://youtube.com/shorts/IXnaBLU_fHQ?feature=share)

### Reproduction

The reproduction is shown in the video.

### Logs

```Text

```

RogueMaster, 2025-04-13:
Unfortunately there are some memory issues with NFC, it is something being worked on and not an ongoing issue on all cards...

closed 2025-04-13 by RogueMaster

DroWez, 2025-04-13:
An unlocked NFC key with a Chameleon behaves the same way. I used to be able to read these keys before.

RogueMaster, 2025-04-15:
See if latest changes have fixed your issue

DroWez, 2025-04-19:
I didn't have time to get to the flipper earlier, but now I've checked it, and everything works as it should.